RobotecAI/ros2-for-unity #12: ROS2UnityComponent - decide on contraint

adamdbrw, 2021-08-28:
Either test multiple components and confirm they work or enforce singleton

kielczykowski-rai, 2021-08-30:
I tested using multiple `ROS2 Unity Component`s in different configurations over Unity project (for example adding components to single or  multiple `GameObjects`).  Everything is working fine inside Unity and ros2 environment. I believe enforcing this component to become singleton will be much more time consuming, than letting every  Unity `GameObject` have it's own  `ROS2 Unity Component` (if it doesn't bring any performance drawback).

adamdbrw, 2021-08-30:
From design principle it is meant as one-of, since it centralizes initialization and destruction, holds all the nodes etc. Multiples only work because we made ros2cs resilient to double-init and so on.

l13517069919, 2022-09-26:
I encountered a similar problem by using unity2021
[image]