RobotecAI/ros2-for-unity #19: RuntimeError: failed to load shared library 'rmw_cyclonedds_cpp.dll' due to LoadLibrary error: 12

chngdickson, 2021-10-04:
My talker and listener is working fine.
Passed the test on ./test.ps1 
However this error pops up when trying to run it on Unity

adamdbrw, 2021-10-04:
Could you please provide logs (Editor.log file)?
I understand you are running a released package on Windows. Are you running standalone or is ros2 installed?
Are you constructing any ROS2 objects before Ros2ForUnity is initialized? There is a couple of branches in validation right now that would make it work if you do, but current master and release packages require you to only construct Publishers, Subscribers and Messages once you initialize the R2FU.

I am mostly guessing here - please provide more context and logs and I will help you make it work.

chngdickson, 2021-10-04:
Thank you for replying so quickly.

Q1 : Are you running standalone or is ros2 installed
ros2 galactic is installed from source.

Q2 : Are you constructing any ros2 objects before Ros2ForUnity is initialized?
I dont understand. I basically created a Unity object and attached the following scripts to them.
[image]

Q3: Log files
Sorry I am not quite sure which log files u require. I hope this is what u are looking for.
[image]

adamdbrw, 2021-10-04:
If you rename your asset (Ros2 directory in Assets) to Ros2ForUnity it may work.

The case is that asset name definition is rather.. static at the moment:

Alternatively, you can set the name to what you prefer in Ros2ForUnity.cs at line 30:
`private static string ros2ForUnityAssetFolderName = "Ros2ForUnity";`

However, this also means that your ros2 installation was not visible (not sourced) by the system, which means you would be using standalone (as in: not requiring ros2 installation) version of Unity R2FU. Which is not a problem, just signalling this to you.

chngdickson, 2021-10-04:
[image]
Unfortunately it still isnt working. I think it's mostly due to the rmw_implementation.

I tried rmw_fastrtps_cpp and rmw_cyclonedds_cpp

No luck thus far

adamdbrw, 2021-10-04:
Can you find rmw_cyclonedds_cpp.dll on your system and tell me where it is? 
Should be two locations: the Ros2ForUnity Asset plugins directory and your ros2 galactic installation folder.

The error is due to plugin loader not being able to find this library. 

Did you follow installation steps in README? I see signs suggesting that perhaps you did it a bit differently.

chngdickson, 2021-10-04:
**rmw locations**
1. C:\Users\Dicks\Documents\unity_projects\New Unity Project\Assets\Ros2ForUnity\Plugins\Windows\x86_64\rmw_cyclonedds_cpp.dll
2. C:\dev\ros2_galactic\install\bin

Here's what is in my path. 
C:\Users\Dicks\Documents\unity_projects\New Unity Project\Assets\Ros2ForUnity\Plugins\Windows\x86_64;C:\Users\Dicks\Documents\unity_projects\New Unity Project\Assets\Ros2ForUnity\Plugins\Windows\x86_64;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.1\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.1\libnvvp;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.0\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.0\libnvvp;C:\Python38\;C:\Python38\Scripts\;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.0\include;C:\tools\cuda\bin;C:\Program Files (x86)\Razer Chroma SDK\bin;C:\Program Files\Razer Chroma SDK\bin;C:\Program Files (x86)\Razer\ChromaBroadcast\bin;C:\Program Files\Razer\ChromaBroadcast\bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Windows\system32\config\systemprofile\AppData\Local\Microsoft\WindowsApps;C:\ProgramData\chocolatey\bin;C:\Program Files\OpenSSL-Win64\bin\;C:\opencv\x64\vc16\bin;C:\Program Files\dotnet\;C:\ProgramData\chocolatey\lib\cunit\lib;C:\ProgramData\chocolatey\lib\tinyxml2\lib;C:\ProgramData\chocolatey\lib\log4cxx\lib;C:\ProgramData\chocolatey\lib\bullet\lib;C:\Program Files\Graphviz\bin;C:\Program Files\Cppcheck;C:\Program Files\Git\cmd;C:\xmllint\bin;C:\Program Files\nodejs\;C:\Program Files\NVIDIA Corporation\Nsight Compute 2020.2.0\;C:\Program Files (x86)\dotnet\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Python38;C:\Python38\Scripts\;C:\Python38\Lib\site-packages\pywin32_system32;C:\Users\Dicks\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\Dicks\AppData\Local\GitHubDesktop\bin;C:\Program Files\heroku\bin;C:\Users\Dicks\.dotnet\tools;C:\Program Files (x86)\GnuWin32\bin;C:\Users\Dicks\AppData\Roaming\npm;C:\Users\Dicks\.dotnet\tools

chngdickson, 2021-10-04:
> Can you find rmw_cyclonedds_cpp.dll on your system and tell me where it is? Should be two locations: the Ros2ForUnity asset Plugins directory and your ros2 galactic installation folder.
> 
> The error is due to plugin loader not being able to find this library.
> 
> Did you follow installation steps in README? I see signs suggesting that perhaps you did it a bit differently.
> 
> Also: try debugging if there is an issue with plugin visibility:
> 
> Add a line under line 132 in Ros2ForUnity.cs to debug what is in 'pluginPath + envPathSep + currentPath', then please check if this path matches your rmw_cyclonedds_cpp.dll location. `Environment.SetEnvironmentVariable(GetEnvPathVariableName(), pluginPath + envPathSep + currentPath);`

I apologize if i was careless. I dont mean to...

adamdbrw, 2021-10-04:
I did not mean that it was careless - we want to support freedom of use in any way you want, it is simply more challenging since there are so many ways! I am determined to make it work for you and will be happy when we succeed!

Is this the Unity debug from the line (or did you check in your system)? 

It looks like everything should work. Your cyclone .dll should be visible by LoadLibrary since it is in the PATH (even twice, for some reason), so I would assume it is a problem with loading one of its dependencies instead. But all dependencies are there in a standalone version of the plugin. This is quite puzzling.

I am aware of the fact that rcutils shared_library.c code cuts the PATH, but it shouldn't be an issue since both ros2 and our module prepend paths to it.  

We didn't run into this issue before. Would you mind checking this .dll dependencies with dependency walker?
https://stackoverflow.com/questions/14361992/dll-load-library-error-code-126

Could you tell me how you installed the Asset (I know that directory name was changed, but anything else perhaps?)

adamdbrw, 2021-10-04:
Quick check - while the module is standalone, it would help to know if your ros2 galactic is working e.g. you can run simple commands after sourcing it? try `ros2 topic list`.

Otherwise please also see if this can help:
https://github.com/ros2/ros2_documentation/issues/488#issuecomment-823672496

Let me know if you are stuck. The other thing we can do is for you to zip and share your project (as I understand it is a demo setup). We can try running it and seeing if it works for us.

chngdickson, 2021-10-04:
Q1 : Is this the Unity debug from the line or system?
It's from Unity.

Q2. Would you mind checking this .dll dependencies with dependency walker?
Am not familiar with dependency walker. However, here are some noticeable errors in **RCUTILS.dll**
[image]

Q3 Quick check - while the module is standalone, it would help to know if your ros2 galactic is working e.g. you can run simple commands after sourcing it? try ros2 topic list.

If u mean sourcing the r2fu\install\local_setup.ps1 , ros2 no longer works on command prompt.

Yeah i might actually consider installing ros2 from https://github.com/ros2/ros2_documentation/issues/488#issuecomment-823672496 

My procedure of installation
a) perform all prerequisites.
b) get_repos, build **ros2cs**
c) run ./test.ps1 to verify build of ros2cs is successful
d) then follow the instructions for building **R2FU** plugin
e) source R2FU and ros2cs
f) Launch Unity and assets

adamdbrw, 2021-10-05:
A few things to try:
1) Could you check the release and import one of our ready '.unitypackage' assets into a new, empty project? Instructions are in the README.
2) If that works, I would ask you to try to build your own plugin again, this time following the ros2-for-unity README (this is how it is meant to be built).
3) Sourcing R2FU or ros2cs is not needed for anything. Perhaps there is a confusing section in the README somewhere for that, can I improve?

Note that if your ros2 doesn't work on command prompt independently of r2fu (which you would probably want to fix), r2fu can still work correctly.

chngdickson, 2021-10-06:
[image]

It worked after importing the ready ".Ros2ForUnity_WindowsGalactic.unitypackage" assets into an empty project.

### Steps to reproduce.
Source all of the local_setups. in powershell
Then import ros2 Galactic unity package
[image]

chngdickson, 2021-10-06:
Note: I haven't made any changes like download foxy. as i was busy ytd. How it performs in the near future. That still isn't known.

closed 2021-10-06 by chngdickson

chngdickson, 2021-10-08:
> No worries, I am happy it started working for you at least in the release for now. I also added clarification to README about not sourcing.

btw do u guys have a discord channel?