RobotecAI/ros2-for-unity #25: ROS2 Subscriber to /image_raw

tkasidakis, 2021-12-03:
Hello.
I wanted to ask if i set to a Unity Script a Subscriber which subscribes to an /image_raw topic of a Robot, how can i see the picture inside the Unity ?

adamdbrw, 2021-12-05:
You could try this method: https://docs.unity3d.com/ScriptReference/Texture2D.SetPixels.html

tkasidakis, 2021-12-05:
Thans a lot for the answer. Maybe i didn't express it in an appropriate way.  For example at the listener example script, we create
private ISubscription<std_msgs.msg.String> chatter_sub.

What is the respective declaration if we want to subscribe at an /image_raw topic and how we take the actual bytes of the pictutre ?

Thanks a lot in advance

adamdbrw, 2022-01-10:
@tkasidakis apologies, I missed your answer.

Adequate declaration would be:
`private ISubscription<sensor_msgs.msg.Image> image_sub;`

The C# messages are generated from IDL/MSG format and are equivalent to https://github.com/ros2/common_interfaces/blob/master/sensor_msgs/msg/Image.msg. Fields in C# are starting with uppercase letters. You can see .cs messages in your build artifacts once you build ros2-for-unity. In case you are working with the release, here is the relevant part of the Image message class:

```
public class Image : MessageInternals, MessageWithHeader
{
  private IntPtr _handle;
  private static readonly DllLoadUtils dllLoadUtils;

  public bool IsDisposed { get { return disposed; } }
  private bool disposed;

  // constant declarations

  // members
  public std_msgs.msg.Header Header { get; set; }

  // Generic interface for all messages with headers
  public void SetHeaderFrame(string frameID)
  {
    Header.Frame_id = frameID;
  }

  public string GetHeaderFrame()
  {
    return Header.Frame_id;
  }

  public void UpdateHeaderTime(int sec, uint nanosec)
  {
    Header.Stamp.Sec = sec;
    Header.Stamp.Nanosec = nanosec;
  }
  public uint Height { get; set; }
  public uint Width { get; set; }
  public System.String Encoding { get; set; }
  public byte Is_bigendian { get; set; }
  public uint Step { get; set; }
  public byte[] Data { get; set; }
```

As you can see, there is Data array that you can access, much like if you were working with the C++ version of the message.

tkasidakis, 2022-01-11:
@adamdbrw  thanks a lot for the answer. I was in a hurry when i posted this question. I have worked extremelly with ros2forUnity and i can say it is really convenient. Congratulations for your worrk. I also managed to enable the QoS setting of ROS2 at the Unity side.

adamdbrw, 2022-01-11:
I am happy it works for you!

closed 2022-01-11 by adamdbrw