RobotecAI/ros2-for-unity #30: Build errors on Windows

AlgoryxJosef, 2022-01-11:
First of all, great work providing this ROS2 UnityPackage! It's exactly what I'm looking for and have already had the pleasure of trying it out some.
I need to be able to use my own cusom message types, and as I understood it, I need to build ros2-for-unity from source to be able to do that (or is there some other way?).

I have attemted to build from source, carefully following the instructions. I have tried both with a pre-built ROS2 (Foxy) that I installed from: https://ms-iot.github.io/ROSOnWindows/GettingStarted/SetupRos2.html and also with a ROS2 (Galactic) that I built from source myself. Both versions work fine as-is. I use Windows.

When I try to build the ros2-for-unity project, I get the same error message regardless of which ROS2 installation I source prior to starting the build. It seems to be related to `generate_cs` not beeing found. I get the following error message: 

```
Traceback (most recent call last):
    File "C:\Users\Admin\git\ros2-for-unity\install\lib\rosidl_generator_cs\rosidl_generator_cs", line 23, in <module>
      from rosidl_generator_cs import generate_cs
  ImportError: cannot import name 'generate_cs' from 'rosidl_generator_cs' (unknown location)
``` 

I have attached the complete build log, if it helps. 
[ROS-for-Unity-build-err-ros2algoryx.md](https://github.com/RobotecAI/ros2-for-unity/files/7846474/ROS-for-Unity-build-err-ros2algoryx.md) 

I have done a little bit of digging my self, and noticed that the path in `rosidl_generator_cs_module` inside src/ros2cs/src/ros2cs/rosidl_generator_cs/bin/rosidl_generator_cs points to a file that does not exists. It points to something like _ros2-for-unity\install\lib\rosidl_generator_cs\__init__.py_ but the file actually seems to be located at _ros2-for-unity\install\lib\site-packages\rosidl_generator_cs\__init__.py_

I tried manually patching this path, and the build now continued a little while longer, but now I get a new error saying :

```
C:\Users\Admin\git\ros2-for-unity\build\rosgraph_msgs\rosidl_generator_cs\rosgraph_msgs\msg\clock.cs(31,10): error CS0246: The type or namespace name 'builtin_interfaces' could not be found (are you missing a using directive or an assembly reference?) [C:\Users\Admin\git\ros2-for-unity\build\rosgraph_msgs\rosgraph_msgs_assembly\rosgraph_msgs_assembly_dotnetcore.csproj] [C:\Users\Admin\git\ros2-for-unity\build\rosgraph_msgs\rosgraph_msgs_assembly.vcxproj]

  Build FAILED.
```

So I'm guessing that there may be some other issues as well. I hope this info helps, I would be really exited being able to use ros2-for-unity with custom messages. Let me know if I can help out, I can test certain things here locally for example if it helps.

Best regards,
Josef

adamdbrw, 2022-01-12:
Thank you for raising the issue, we will look into it!

AlgoryxJosef, 2022-01-19:
I'm closing this since this is a duplicate of: https://github.com/RobotecAI/ros2cs/issues/14

closed 2022-01-19 by AlgoryxJosef