RobotecAI/ros2-for-unity #36: Generates ROS1 messages with ROS2 setting

philipholler, 2022-03-03:
**Describe the bug**
I have created a local custom package with some msgs, srvs, and actions. I sourced it correctly, and it runs fine.

However, when i want to generate .cs files to use inside unity (going to Robotics -> Generate ROS Messages -> Build n msgs) the messages are generated alright, but they seem to be generated for ROS1. This became clear when i tried to use the class as type parameter to the ROS2.ISubscription. I then get the error message "The type 'RosMessageTypes.MyPkg.CustomMsg' must be convertible to 'ROS2.Message'".

I have set the protocol to ROS2 under Robotics -> ROS Settings -> Protocol.

I would expect the generated classes to extend the ROS2.Message class, but instead they extend a Message class from "Unity.Robotics.ROSTCPConnector.MessageGeneration".

**Desktop:**
 - OS: Ubuntu 20.04
 - ros2 distro: Galactic
 - ros2-for-unity version: Release 1.1.0
 - ros2cs version: Unsure?
 - ros2 dds middleware: Default settings, i have not changed anything
 - ros2 environment setup: Local pc

adamdbrw, 2022-03-03:
By the way, we do not support sevices and actions in ros2cs yet - it would be a nice contribution to add these!

adamdbrw, 2022-03-07:
@philipholler does my explanation clarify things? Were you able to run your sim with custom messages?

philipholler, 2022-03-07:
Yes, it explained a lot, thank you!

Since i need services, i switched to using the Unity Robotics Hub.

closed 2022-03-07 by adamdbrw