RobotecAI/ros2-for-unity #39: Cannot write some variables in ROS2 messages.

Whypac, 2022-03-20:
When a variable's name is the same as the message's type name, Unity will send a compile error.

For example:

rosgraph_msgs.msg.Clock msg = new rosgraph_msgs.msg.Clock();
builtin_interfaces.msg.Time time_msg = new builtin_interfaces.msg.Time();
time_msg.Sec = sec;
time_msg.Nanosec = nanosec;
msg.Clock = time_msg;

This will cause an error as such:
error CS1061: 'Clock' does not contain a definition for 'Clock' and no accessible extension method 'Clock' accepting a first argument of type 'Clock' could be found (are you missing a using directive or an assembly reference?)

Another example:
nmea_msgs.msg.Sentence msg = new nmea_msgs.msg.Sentence();
msg.Header.Stamp.Sec = sec;
msg.Header.Stamp.Nanosec = nanosec;
msg.Sentence = (string)gps.gpgga;

The error:
error CS1061: 'Sentence' does not contain a definition for 'Sentence' and no accessible extension method 'Sentence' accepting a first argument of type 'Sentence' could be found (are you missing a using directive or an assembly reference?)

In the second example, writing the header isn't causing any trouble, so I wonder what causes the problem. I apologize if I have made some simple mistakes.
Your help will be appreciated.

adamdbrw, 2022-03-25:
@Whypac 
This is handled automatically. If conflicting name is found, an underscore is appended. 
For example, here is a fragment of the generated clock.cs file:
`public builtin_interfaces.msg.Time Clock_ { get; set; }`

This is not documented for the user, we will add some clarification. This should help you.

Whypac, 2022-03-26:
@adamdbrw 
That really helps. Thanks. I'm closing this issue.

closed 2022-03-26 by Whypac