RobotecAI/ros2-for-unity #42: RuntimeError: error not set, at /tmp/binarydeb/ros-foxy-rcl-1.1.11/src/rcl/node.c:276

Raphtor, 2022-04-08:
**Describe the bug**
I am getting a runtime error when running the example. I cannot see either node with `ros2 topic list` or `ros2 node list`.

The error

```
RuntimeError: error not set, at /tmp/binarydeb/ros-foxy-rcl-1.1.11/src/rcl/node.c:276
ROS2.Utils.CheckReturnEnum (System.Int32 ret) (at /home/pjaroszek/projects/robotec/tasks/github/ros2-for-unity/release-1.1/ros2-for-unity-foxy/src/ros2cs/src/ros2cs/ros2cs_core/utils/Utils.cs:37)
ROS2.Node..ctor (System.String nodeName, ROS2.rcl_context_t& context) (at /home/pjaroszek/projects/robotec/tasks/github/ros2-for-unity/release-1.1/ros2-for-unity-foxy/src/ros2cs/src/ros2cs/ros2cs_core/Node.cs:61)
ROS2.Ros2cs.CreateNode (System.String nodeName) (at /home/pjaroszek/projects/robotec/tasks/github/ros2-for-unity/release-1.1/ros2-for-unity-foxy/src/ros2cs/src/ros2cs/ros2cs_core/Ros2cs.cs:130)
ROS2.ROS2Node..ctor (System.String unityROS2NodeName) (at Assets/Ros2ForUnity/Scripts/ROS2Node.cs:38)
ROS2.ROS2UnityComponent.CreateNode (System.String name) (at Assets/Ros2ForUnity/Scripts/ROS2UnityComponent.cs:86)
ROS2.ROS2ListenerExample.Update () (at Assets/Ros2ForUnity/Scripts/ROS2ListenerExample.cs:39)
```

Note: `/home/pjaroszek/projects/robotec/tasks/github/ros2-for-unity/release-1.1/ros2-for-unity-foxy/src/ros2cs/src/ros2cs/ros2cs_core/Node.cs:61` is not a path on my system. @pijaro it seems that your paths are hard coded somewhere in the release?

**To Reproduce**
Steps to reproduce the behavior:
1. Download the latest release (1.1.0)
1. Source ROS distro
2. Launch editor
3. Create two empty objects, add a ROS Unity Component to each of them
4. Add the talker example script to one
5. Add the listener example script to the other
6. Hit play

**Expected behavior**
The nodes talk to each other, and I can see them with `ros2 topic list` and `ros2 node list`

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - ros2 distro: foxy
 - ros2-for-unity version 1.1.0 release
 - ros2cs version: unsure, I think 1.1
 - ros2 dds middleware: cyclonedds
 - ros2 environment setup : single pc

**Additional context**
N/A

pijaro, 2022-04-11:
Hello, 

no, there are no hardcoded paths and this has nothing to do with the error. The paths you are getting come from debug symbols which are also included in the unity package.

As for the issue, it looks like there is a problem with creating ros2 node. Could you provide a full editor log?

Also could you:
-  double-check if you are able to publish and receive some dummy messages in the terminal (`ros2 topic pub ...`, `ros2 topic echo ...`),
- try adding `<PROJECT_DIR>Assets/Ros2ForUnity/Plugins/Linux/x86_64` to your `LD_LIBRARY_PATH` before running editor,
- double check if the `foxy` is the only sourced ROS2 version, maybe there is `galactic` in your `~/.profile`/`~/.bashrc` that you are not aware of? (I see that you wrote that you are using `foxy`, but this error looks like the ros versions mismatch),
- tell me if this error also happens when using `rmw_fastrtps_cpp` as a DDS middleware?

adamdbrw, 2022-05-09:
@Raphtor any luck with the error following @pijaro explainations?

Raphtor, 2022-05-18:
Yes, I tried all of those. I ended up just using the ROS2 solution supported by Unity via their TCP connector.

closed 2022-05-18 by adamdbrw