RobotecAI/ros2-for-unity #44: ROS2TalkerExample: Object reference not set to an instance of an object

joe28965, 2022-04-29:
**Describe the bug**
When I try the README "Usage" step 9: 
Once you start the project in Unity, you should be able to see two nodes talking with each other in Unity Editor's console or use ros2 node list and ros2 topic echo /chatter to verify ros2 communication.

I get the error:
`Object reference not set to an instance of an object ROS2.ROS2TalkerExample.Update() (at Assets/Ros2ForUnity/Scripts/ROS2TalkerExample.cs:38)`
With line 38 being:
```
if (ros2Unity.Ok())
```

I get the same error with the listener, except line 37
```
if (ros2Node == null && ros2Unity.Ok())
```

That makes me guess that something goes wrong in
```
void Start()
{
    ros2Unity = GetComponent<ROS2UnityComponent>();
}
```

Except I don't know what.

**Expected behavior**
I'd expect to see the talker and listener to work.

**Desktop (please complete the following information):**
 - OS: Ubuntu20.04
 - ros2 distro Foxy
 - ros2-for-unity version: master branch 90f009aeb4ac47e16fdc1ea4152b0f350d51e749
 - ros2 environment setup: single pc

pijaro, 2022-05-04:
Hello @joe28965 

Do you have `ROS2UnityComponent` attached to the same game object as the `ROS2TalkerExample`? Examples (talker and listener) require `ROS2UnityComponent` to be in the same game object.

joe28965, 2022-05-04:
Thank you so much, that was indeed the problem.

However, that does raise a question. Maybe I'm not interpreting it correctly, but if I look at the README it says the following:

> 6. Create a top-level object containing ROS2UnityComponent.cs. This is the central Monobehavior for ROS2 For Unity that manages all the nodes. Refer to class documentation for details.
> 7. Add example script *to any object* in the hierarchy tab, e.g. by dragging ROS2TalkerExample.cs to the object in the inspector tab.
> 8. *Select another object* in the hierarchy tab and add repeat the previous step using ROS2ListenerExample.cs.

The way I'm reading it, I don't need to add it to the same game object. I will admit I have basically no Unity experience, so it could very well be me. However, I'd love to hear your opinion.

pijaro, 2022-05-09:
Glad it helped. 

You are right that the current version of README isn't detailed enough. We are going to update it in the upcoming release.

closed 2022-05-09 by pijaro