RobotecAI/ros2-for-unity #46: spawn gameobject in unity based on real world coordinates

kkirby04, 2022-07-15:
hi , i am using computer vision to detect peoples position and would like to be able to spawn their positions in unity. so far i have the publisher publishing these corrdinates of the peoepl in the real world to unity subscriber node. but now i am unsure as to how to wirte in the ros2 namespace language you guys have made to spawn these people at the subscribed positions. any help would be great.
many thanks,
kieran

adamdbrw, 2022-08-18:
@kkirby04 I got to this quite late, hope you managed to find a solution. You would need to implement a solution in the application layer, this project does not handle that. There are conversions you could use for coordinates themselves, but implementing a Map component (with configurable fix) is a good place to start.

pijaro, 2022-11-02:
I believe that this is Unity specific issue, and it is best to look for an answer on [unity answers](https://answers.unity.com/index.html) or inside [Unity documentation](https://docs.unity3d.com/) ([Object.Instantiate](https://docs.unity3d.com/ScriptReference/Object.Instantiate.html) might be something that you are looking for).

closed 2022-11-02 by pijaro