RobotecAI/ros2-for-unity #48: include ros services and actions

TKaluza, 2022-09-04:
Would be great to have also [ros2 services ](https://docs.ros.org/en/humble/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Services/Understanding-ROS2-Services.html) or [ros2 actions](https://docs.ros.org/en/humble/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Actions/Understanding-ROS2-Actions.html) available! 
Thank you for your create unity asset!

adamdbrw, 2022-09-05:
@TKaluza I agree these would be great features! We have an issue for services here (in associated repository): https://github.com/RobotecAI/ros2cs/issues/1 - it would be a valuable PR! 
Supporting services is something we might still do this year, but since we are not using them ourselves, it would be great to have a contribution.

adamdbrw, 2023-02-14:
Services are now supported, I will rename the issue to reflect that

samiamlabs, 2023-03-08:
Is anyone currently working on adding support for actions?
I have some time over and can give it a shot if is is not already in progress :)

adamdbrw, 2023-03-08:
@samiamlabs No work that I know of is being done on actions at the moment, would be great if you did it!