RobotecAI/ros2-for-unity #49: dependency of custom message could not be found

Deric-W, 2022-09-29:
**Describe the bug**
The following error occurs while building a part of  [this](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Custom-ROS2-Interfaces.html) package:

```
C:\dev\ros2-for-unity\build\ros2_custom_msgs\rosidl_generator_cs\ros2_custom_msgs\msg\sphere.cs(31,10): error CS0246: The type or namespace name 'geometry_msgs' could not be found (are you missing a using directive or an assembly reference?) [C:\dev\ros2-for-unity\build\ros2_custom_msgs\ros2_custom_msgs_assembly\ros2_custom_msgs_assembly_dotnetcore.csproj] [C:\dev\ros2-for-unity\build\ros2_custom_msgs\ros2_custom_msgs_assembly.vcxproj]
```

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the tutorial (omit `Num.msg` and `AddThreeInts.srv` in `msg`and `CMakeLists.txt`)
2. add the following to `ros2_for_unity_custom_messages.repos`:

```
    src/ros2cs/custom_messages/<package_name>:
        type: git
        url: <url>
        version: humble
```

3. create a Windows build
4. watch it fail to build the custom package

**Expected behavior**
The build succeeds

**Desktop:**
 - OS: Windows 10
 - ros2 distro: humble
 - ros2-for-unity version: 1.2.0
 - ros2cs version [e.g. git-sha 2a45785b080fffb3ae5e2f645e976a69698810f0]
 - ros2 dds middleware: fastdds
 - ros2 environment setup: single pc

**Fix**
The build succeeds when adding the following to `ros2_for_unity_custom_messages.repos`:

```
    src/ros2cs/src/ros2/<package_name>:
        type: git
        url: <url>
        version: humble
```

pijaro, 2022-11-02:
I tried to reproduce the issue but without success.

I created a custom message package from the mentioned tutorial ([pijaro/tutorial_interfaces](https://github.com/pijaro/tutorial_interfaces)) and set it up in a custom messages file `ros2_for_unity_custom_messages.repos`:

```
# NOTE: Use this file if you want to build with custom messages that reside in a separate remote repo.
# NOTE: use the following format

repositories:
#  src/ros2cs/custom_messages/<package_name>:
#    type: git
#    url: <repo_url>
#    version: <repo_branch>
#  custom_messages/<package2_name>:
#    ...
#    ...

    src/ros2cs/custom_messages/tutorial_interfaces:
        type: git
        url: git@github.com:pijaro/tutorial_interfaces.git
        version: main
```

then `.\pull_repositories.ps1` and `.\build.ps1`. Build succeeded without errors.

Did you check the [Windows important notices](https://github.com/RobotecAI/ros2cs/blob/master/README-WINDOWS.md#important-notices)? A lot of strange errors are most often caused by Windows path restrictions.

Also, It is always a good idea to clear the `build` and `install` directories if the workspace dir changed substantially.

A full build log could be useful as well. Maybe the package and repos are correct, but the problem is that the `geometry_msgs` did not build successfully.

Deric-W, 2022-11-15:
Sorry for the long delay, I cleared the `build` and `install` directories and it worked, sorry for bothering you.

closed 2022-11-15 by Deric-W