RobotecAI/ros2-for-unity #52: Does ROS2-For-Unity work on IOS?

RubenCarreira99, 2022-10-12:
Hi everyone,

This issue is not really an issue, but a question instead. I don´t really know where else I should ask this. I hope you don´t mind if I post it here.

I'm developing an IOS app to control a turtlebot3 wafflePI through ROS2. It is also meant to let users control the waffle's digital twin in a simulated environment using Unity. As I've been working with ROS on an Ubuntu VM, I decided to install Unity on it and then managed to integrate ros2-for-unity package. At the moment, I can control the waffle's model using my keyboard.

The objective is to have buttons on the iPad's screen publishing values to cmd_vel topic hence making the wafflePI move.

 Now the question is - Will the ros2-for-unity package still work when I integrate the unity project with my IOS app? 
 
 Hope I was clear enough. It is kind of tricky for me to explain.

pijaro, 2022-11-02:
Hello,

unfortunately, we are not supporting iOS currently, and we did not try to run it on iOS at any point. However, I'm almost certain that it will not run out of the box because of the difference in the .net SDK.

adamdbrw, 2022-12-05:
Closing the issue. If you would like support for IOS, please make a feature request (A dedicated issue)

closed 2022-12-05 by adamdbrw