RobotecAI/ros2-for-unity #53: UnsatisfiedLinkError: libsensor_msgs__rosidl_generator_c.so

tigremevlude, 2022-11-01:
**Describe the bug**
I got UnsatisfiedLinkError: libsensor_msgs__rosidl_generator_c.so. Is the problem related to ros2 or [ros2-for-unity](https://github.com/RobotecAI/ros2-for-unity)?  

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. add  `image_pub = ros2Node.CreatePublisher<sensor_msgs.msg.CompressedImage>("image"); ` to talker.cs script
2. inside the lateUpdate() method 
`
           

            Texture2D snapshot = new Texture2D(resWidth, resHeight, TextureFormat.RGB24, false);
            snapCam.Render();
            RenderTexture.active = snapCam.targetTexture;
            snapshot.ReadPixels(new Rect(0,0,resWidth, resHeight), 0, 0);
            byte[] bytes = snapshot.EncodeToPNG();

            sensor_msgs.msg.CompressedImage ImageMsg = new sensor_msgs.msg.CompressedImage();

            ImageMsg.Data = bytes;
            ImageMsg.Format = "png";
            image_pub.Publish(ImageMsg);`
4. add a camera and link the render texture to camera
5. Link the talker.cs script and Unity Component.cs script to camera 
6. Run the Unity simulation

**Expected behavior**
First, I did not receive any error but after some time Unity freezed and I got the link error because of the sensor_msgs. 

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu20.04]
 - ros2 distro [foxy]
 - ros2-for-unity version [latest]
 - ros2cs version [latest]
 - ros2 dds middleware [e.g. cyclonedds, fastdds]
 - ros2 environment setup [single pc]

pijaro, 2022-11-02:
I see that you are running the 1.1 version. Can you try the [latest 1.2 release](https://github.com/RobotecAI/ros2-for-unity/releases/tag/1.2.0)?

tgrmev, 2022-11-02:
I also tried to ros2-for-unity-1.2.0 version. When I tried to run the unity with this version,  I got this error : 

> NullReferenceException: Object reference not set to an instance of an object
ROS2.ROS2UnityComponent.OnApplicationQuit () (at Assets/Ros2ForUnity/Scripts/ROS2UnityComponent.cs:160)

Also to give information, when I build the version 1.2.0 verison end of the building process I received this message :  

> Summary: 46 packages finished [7min 16s]
  19 packages had stderr output: action_msgs actionlib_msgs builtin_interfaces diagnostic_msgs geometry_msgs lifecycle_msgs nav_msgs rcl_interfaces rosgraph_msgs sensor_msgs shape_msgs statistics_msgs std_msgs stereo_msgs test_msgs tf2_msgs trajectory_msgs unique_identifier_msgs visualization_msgs
cp: cannot stat '/home/mev/UnityProjects/Unity-Ros2/ros2-for-unity-1.2.0_new/src/Ros2ForUnity/metadata_ros2cs.xml': No such file or directory
cp: cannot stat '/home/mev/UnityProjects/Unity-Ros2/ros2-for-unity-1.2.0_new/src/Ros2ForUnity/metadata_ros2cs.xml': No such file or directory

pijaro, 2022-11-07:
@tgrmev 

Looks like the meta files were not generated properly. 

Could you run 
```bash
python3 ./src/scripts/metadata_generator.py
```
from the `ros2-for-unity` dir and see if `src/Ros2ForUnity/metadata_ros2cs.xml` and `src/Ros2ForUnity/metadata_ros2_for_unity.xml` files were generated?

adamdbrw, 2022-12-05:
@tgrmev any luck following @pijaro advice?

tgrmev, 2022-12-06:
hello, yes now i do not receive this error. Thanks.

closed 2022-12-06 by tgrmev

ynyBonfennil, 2023-02-06:
Since this issue is already closed, it might be pointless to write a comment, but I got a similar error `UnsatisfiedLinkError: libstd_msgs__rosidl_typesupport_c.so` in my personal project with ros2-for-unity v1.2.

The reason was because I instantiated `std_msgs.msg.String` before instantiating `ROS2UnityCore` (I created the message at Awake() but I instantiated `ROS2UnityCore` in Start()).

It seems like internal class `ROS2ForUnity` plays an important role for loading libraries but it won't be instantiated unless creating `ROS2UnityCore` first.