RobotecAI/ros2-for-unity #54: don't know how to custom message

junmeng6025, 2022-11-09:
I tried to custom the message in `ros2_for_unity_custom_messages.repos`, like this:
```
# NOTE: Use this file if you want to build with custom messages that reside in a separate remote repo.
# NOTE: use the following format

repositories:
  src/ros2cs/custom_messages/<package_name>:
    type: git
    url: <repo_url>
    version: <repo_branch>
  custom_messages/<package2_name>:
    "Hello World"

```
and then ran `./pull_repositories.sh` in root directory, got this:
```bash
=========================================
* Pulling ros2cs repository:
=== ./src/ros2cs/ (git) ===

HEAD is now at 7e3d179 msgs.cs.em platform check based on selected csbuildtool

=========================================
Pulling custom repositories:
Input data is not valid format: string indices must be integers

=========================================
Pulling ros2cs dependencies:
Detected ROS2 galactic. Getting required repos from 'ros2_galactic.repos'
........

```
It seems that my message in wrong syntax. How should I edit the message?

pijaro, 2022-11-10:
Hello,

to use custom messages in `ros2-for-unity` you have to create a separate package with your messages. You can find the instructions on how to do it here: [Creating custom msg and srv files](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Custom-ROS2-Interfaces.html).

Then, having such a custom package, you need to put it inside `src/ros2cs` directory or use a `.repos` file. 

If you are hosting your package on a git repository, you can list it in `ros2_for_unity_custom_messages.repos` file.  `ros2_for_unity_custom_messages.repos` doesn't contain the custom messages definitions directly. It holds the information where the build system can find the repository for that package. You can't define messages in `.repo` file as you tried to do (see [Custom messages](https://github.com/RobotecAI/ros2-for-unity#custom-messages) in a readme).

adamdbrw, 2022-12-05:
@junmeng6025 were you able to work with custom message following the instructions? If so, could you close the issue?