RobotecAI/ros2-for-unity #55: RuntimeError - when using custom messages

kimjjs100, 2022-12-06:
Hi, there?
I have some problem when using my custom msgs.

**Describe the bug**
- After inserting "custom msgs" in `src/ros2cs`, I executed `./build.sh` and imported the Asset into Unity.  
  - There are some `stderr output`, but it was confirmed that the build was completed. 
  - [image]

- Also, it was confirmed that there is no problem in executing `ROS2TalkerExample.cs` and `ROS2ListenerExample.cs`. (In 'std_msgs', it was confirmed that the topic was published normally.)

- However, when I changed `std_msgs` to `custom_msgs` in the example template, the following error occurred.
  - (Original Code)
  - [image]

  - (The error)
  - [image]

```
RuntimeError: type support not from this implementation, at /tmp/binarydeb/ros-foxy-rmw-fastrtps-cpp-1.3.1/src/publisher.cpp:86, at /tmp/binarydeb/ros-foxy-rcl-1.1.14/src/rcl/publisher.c:180
ROS2.Utils.CheckReturnEnum (System.Int32 ret) (at <e0a206ab668a41a18192d5dc278b96ab>:0)
ROS2.Publisher`1[T]..ctor (System.String pubTopic, ROS2.Node node, ROS2.QualityOfServiceProfile qos) (at <e0a206ab668a41a18192d5dc278b96ab>:0)
ROS2.Node.CreatePublisher[T] (System.String topic, ROS2.QualityOfServiceProfile qos) (at <e0a206ab668a41a18192d5dc278b96ab>:0)
ROS2.ROS2Node.CreatePublisher[T] (System.String topicName, ROS2.QualityOfServiceProfile qos) (at Assets/Ros2ForUnity/Scripts/ROS2Node.cs:75)
ROS2.ToKSOE.Update () (at Assets/ToKSOE.cs:44)
```

- I think the `Null Reference` problem seems to be something that won't happen if only runtime errors are resolved.

**Desktop (please complete the following information):**
 - OS : `Ubuntu20.04`
 - ros2 distro : `foxy`
 - ros2-for-unity version : `1.2.0 Release`
 - ros2cs version : `1.2.0 Release`
 - ros2 dds middleware : `fast-dds`
 - ros2 environment setup : `single pc`, `local network`
 - Unity editor version : `2020.3.26f1`

I need your help a lot !! How should I handle this issue?

pijaro, 2022-12-12:
Is it possible that you could share the contents of your custom messages (package)? I could try to reproduce it locally.

Also, did you make sure you cleared your Unity plugins folder and {build/install} before copying and building new libraries? The error looks like some libraries might be from different versions :thinking:

pijaro, 2023-01-23:
Did you manage to find the solution to your problem?

kimjjs100, 2023-02-10:
I'm sorry.. I'm not working on solving this problem anymore.
When it's resolved, I'll raise it as an issue again.

closed 2023-02-10 by kimjjs100

dsponer, 2023-05-20:
Hello
I got the same problem with my custom messages
I have successfully collected the messages and uploaded them to the project

This is my custom message (Light.msg)

uint8 brightness
uint8 channel