RobotecAI/ros2-for-unity #59: running my Unity software on Windows and ROS2 foxy on Linux(virtualbox)

jitliang, 2023-01-19:
Hi, I have been trying to connect my Unity project to ROS2 by following the tutorial here . However, I have been unsuccessful. Currently, I am running my Unity software on Windows and ROS2 foxy on Linux(virtualbox). This means both Unity and ROS2 are running in the same PC, however under different OS. May I know which pre-built releases should I download? Under such setup, can your module still work? Thanks!

**Describe the bug**

/chatter is not detected at ROS2.

**Desktop (please complete the following information):**

OS : Ubuntu20.04
ros2 distro : foxy
ros2-for-unity version : 1.2.0 Release
ros2cs version : 1.2.0 Release
ros2 dds middleware : fast-dds
ros2 environment setup : single pc, local network
Unity editor version : 2020.3.26f1

cuhde, 2023-02-13:
I have the same issue with a separate PC running Ubuntu 20.04 and ROS2 Foxy. ros2-for-unity 1.2.0 release for Win10 x64 in Unity 2021.3.18f 
I can receive chatter in both directions using bash and cmd. I can even receive chatter from the Win cmd into Unity. But As soon as I publish a String topic on the ubuntu machine, I don't receive anything.

cuhde, 2023-02-13:
Okay, in my case it was the windows firewall. By disabling it I am now able to receive data from the Ubuntu PC in the local network.

adamdbrw, 2023-02-14:
I am glad this worked for you, closing the issue!

closed 2023-02-14 by adamdbrw