RobotecAI/ros2-for-unity #61: Memory Leak

patricia241, 2023-02-13:
When I do a suscriptor in Unity and open task manager of Windows I can see how memory usage increases uncontrollably over time. With usual messages such as a script, the computer does not become saturated, but when it receives images it does collapse. I have had this same experience in other projects such as the Unity Robotics Hub.

**To Reproduce**
Create an empty object in Unity Hierarchy and add ROS2UnityComponent.cs, ROS2TalkerExample and ROS2ListenerExample. Play de game on Unity Editor and see how it increases the value of the memory of the process that executes the project on WIndows task manager. 

I think you should add the parameter of how many messages you want in the queue when declaring the subscriber and delete all messages that are not required, just like the original ros2 subscriber does.

patricia241, 2023-02-14:
@adamdbrw I created a subscriber on Unity, my publisher is the libgazebo_ros_camera.so plugin. Publisher is in a Virtual Machine with Ubuntu 20.04 and ros2 foxy. I'm new in C# and with Unity. The intention of my project is show the image of gazebo camera on Oculus Quest 2. Actually, seems  memory increase but over the time is released.  Even so, with the Oculus connected, the Unity editor crashes. I don't know why.

Sorry for this issue it was my fault. 

My Unity script is the following in case you could help me to know what I do wrong so that the editor crashes: 

~~~
using ROS2;
using UnityEngine;
using UnityEngine.UI;

public class CamScript : MonoBehaviour
{
    Texture2D texRos;
    public RawImage display;

    private ROS2UnityComponent ros2Unity;
    private ROS2Node ros2Node;
    private ISubscription<sensor_msgs.msg.Image> image_sub;

    private sensor_msgs.msg.Image img_msg = null;

    void Start()
    {
        ros2Unity = GetComponent<ROS2UnityComponent>();
    }

    void Update()
    {
        if (ros2Node == null && ros2Unity.Ok())
        {
            ros2Node = ros2Unity.CreateNode("ROS2UnityCameraNode");
            QualityOfServiceProfile qos = new QualityOfServiceProfile(QosPresetProfile.SENSOR_DATA);
            image_sub = ros2Node.CreateSubscription<sensor_msgs.msg.Image>(
              "/global_camera/image_raw", msg => { this.img_msg = msg; }, qos);
        }

        if (img_msg != null)
        {
            texRos = new Texture2D((int)img_msg.Width, (int)img_msg.Height, TextureFormat.RGB24, false);
            texRos.LoadRawTextureData(img_msg.Data);

            texRos.Apply();
            display.texture = texRos;
        }
    }
   
}
~~~
Thanks for your early response.

telemething, 2023-02-15:
Change the code to create texRos only once.

adamdbrw, 2023-02-15:
@patricia241 thank you for sharing a snipped, it is much more clear now!
- It seems you could be running into synchronization issues with img_msg. Subscriptions are processed in a separate thread. You should consider adding a mutex.
- Using new for a larger memory size in Update will cause performance issues (allocations, deallocations, spikes when GC is run). As @telemething points out, it is better to create the texture once and, in addition only modify it when an actual new message arrived (you can check if you got a new img_msg).
- Your memory footprint is increased by this allocation in Update(), and could depend on your frame rate, GC execution and so on. This does not have mean that there is a memory leak in R2FU. 
- This case (receiving images in Unity3D, as opposed to publishing them from Unity3D, as would be for camera sensor data) is not tested in our cases and not optimized for performance. I am not saying there could be dragons, but there is for sure some work that could be done to optimize read performance in ros2cs in such case.

I am closing the issue since it does not show a memory leak in R2FU.

closed 2023-02-15 by adamdbrw