RobotecAI/ros2-for-unity #62: How to create custom message and use it with ros2-for-unity repository in Unity?

tgrmev, 2023-02-16:
Hello,
I use your repository to receive the camera information from Unity as a ros2 message. I transfer the cam_image, cam_XYZ, cam_RPY, and cam_FOV messages seperately.  I want to create a new custom **cam_info** message that contains cam_XYZ, cam_RPY, and cam_FOV information. How can I do that with your repository? For example, I use formal ros2 messages like this **private IPublisher<sensor_msgs.msg.CompressedImage> image_pub;**

adamdbrw, 2023-02-16:
@tgrmev please refer to https://github.com/RobotecAI/ros2-for-unity#custom-messages section of the README. It is enough to build the messages. Also, you might take a look at ros2cs part of the instruction: https://github.com/RobotecAI/ros2cs#generating-custom-messages. 

In short, it is enough to create the ros2 message package, put it in the specified folder and build the R2FU. You will have new plugins for your Unity project.

adamdbrw, 2023-02-16:
If you find some things not clear enough, please feel welcome to submit a pull request with additional documentation

adamdbrw, 2023-03-08:
@tgrmev closing this issue

closed 2023-03-08 by adamdbrw