RobotecAI/ros2-for-unity #63: CMake Error - Target Names - Windows 10 - Humble

Mgenenger, 2023-02-27:
When running the ./build-ps1 command, I get the following error:

[image]

This happens on package 16/47. I tried it with CMake version 3.24.0 and 3.25.0. For this I have the Ament_cmake version 1.3.2 and Windows SDK version 10.0.19041.0 on Windows 10.
Do you have a quick idea or a workaround? Is the whole thing possibly a Windows problem and on Ubuntu the error does not exist? Thanks for your help.

adamdbrw, 2023-03-17:
We were not able to reproduce the bug when using Windows 10.
This could very likely be platform-specific, it is hard to tell.
Do you still have the problem?

pijaro, 2023-03-19:
To double-check, have you followed the important notices for Windows? 
https://github.com/RobotecAI/ros2cs/blob/master/README-WINDOWS.md#important-notices

From my experience, most of the Windows issues are related to insufficient permissions (run PowerShell with admin rights) or path lengths (change the limit in the Windows registry).

Mgenenger, 2023-03-20:
Hey, thank you very much for the feedback. I found a way around the error with ROS noetic and Ros#. There were no difficulties. 
I ran PowerShell as admin as well as enabled path length. 
For my project ROS is sufficient and I don't necessarily need to work with ROS2, therefore you can neglect the problem for now. I hope that this error is really only with me.

pijaro, 2023-03-20:
:+1: I think that we can close it then.

closed 2023-03-20 by pijaro