RobotecAI/ros2-for-unity #64: Release 1.2.0 did not work on my PC

samiamlabs, 2023-03-08:
**Describe the bug**
[1.2.0](https://github.com/RobotecAI/ros2-for-unity/releases/tag/1.2.0) does not seem to work for me on Linux 22.04 and Unity 2021.3.2 (LTS).
It looks like it was an overlay release so I installed humble using apt-get on my computer and started the editor from a terminal where I already sourced ROS2 humble.

**To Reproduce**
Steps to reproduce the behavior:
1. Start with a computer with Ubuntu 22.04
2. Make a fresh project in Unity 2021.3.2
3. Download [Ros2ForUnity_humble_ubuntu2204.zip](https://github.com/RobotecAI/ros2-for-unity/releases/download/1.2.0/Ros2ForUnity_humble_ubuntu2204.zip)
4. Open a terminal and run `source /opt/ros/humble/setup.bash`
5. Open the Unity editor in the same terminal by writing, for example, `Unity/Hub/Editor/2021.3.20f1/Editor` 
6. Add a ROS2 Unity Component script to an emtpy GameObject.

**Expected behavior**
No errors

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu22.04, Windows 10
 - ros2 distro humble
 - ros2-for-unity version 1.2.0
 - ros2cs version ?
 - ros2 dds middleware fastdds
 - ros2 environment setup single pc

pijaro, 2023-03-08:
Can you verify that the `ROS_DISTRO` is set to `humble` and the `/opt/ros/humble/lib` is in `LD_LIBRARY_PATH` after sourcing the ROS 2?

```bash
env | grep ROS_DISTRO
env | grep LD_LIBRARY_PATH
```

samiamlabs, 2023-03-08:
```
sam@sam-blade:~$ env | grep ROS_DISTRO
ROS_DISTRO=humble

sam@sam-blade:~$ env | grep LD_LIBRARY_PATH
LD_LIBRARY_PATH=/opt/ros/humble/opt/rviz_ogre_vendor/lib:/opt/ros/humble/lib/x86_64-linux-gnu:/opt/ros/humble/lib
```

zp-yang, 2023-03-14:
How did you import the .zip package to unity? I tried to renaming it .unitypackage but got an error for couldn't decompress. I tried decompressing the zip and drag the two folders into an empty project still doesn't seem to work.

adamdbrw, 2023-03-17:
As a quick solution, you might look for the check that outputs this message and disable the check. If the check is invalid and not a symptom of some mismatch, this should be enough. We will be looking into it next week.

pijaro, 2023-03-19:
As @adamdbrw mentioned, try to check the versions which are detected by `Ros2ForUnity` by adding debug log to `Ros2ForUnity/Scripts/ROS2ForUnity.cs` somewhere in the [`if` statement in line 180](https://github.com/RobotecAI/ros2-for-unity/blob/1.2.0/src/Ros2ForUnity/Scripts/ROS2ForUnity.cs#L180-L185):

```c#
Debug.Log("Sourced: " + ros2SourcedCodename + ". From metadata: " + ros2FromRos2csMetadata);
```

Logged ROS 2 versions should be the same and not empty. If it is not the case, then there might be an issue with path resolution.

Can you also make sure that `Ros2ForUnity` folder lives inside project `Assets`?:
```bash
Assets/
└── Ros2ForUnity
    ├── Plugins
    │   └── Linux
    │       └── x86_64
    └── Scripts
        └── Time
```

rahulswa08, 2023-05-20:
Same happened with me then I tried running the Unity from terminal as shown below and it worked. 

https://github.com/RobotecAI/ros2-for-unity/issues/32#issuecomment-1023981776