RobotecAI/ros2-for-unity #66: Some variables are read-only in ros2cs message class

chenjunnn, 2023-03-17:
**Describe the bug**
I want to publish camera info from Unity to ROS2, then I found the `K` variable in `sensor_msgs.msg.CameraInfo` is read-only and there is no constructor which can change the value of `K`. Also the `P` and `R` are read-only, too.

``` csharp
public double[] P { get; }
public double[] R { get; }
public double[] K { get; }
```

The original type in raw sensor_msgs CameraInfo message definition is float64 array with specified length of 9, and all the arrays of specified length in csharp are read-only.
```
float64[9]  K # 3x3 row-major matrix
```

So is there a solution to change the value of `K` in csharp script?

adamdbrw, 2023-03-17:
I believe that having the getter for the array only means you can't set the entire array, but you can set them by index. Have you tried P[0] = x?

chenjunnn, 2023-03-17:
@adamdbrw Thanks for your help! I have tried setting the value by index last time but I forget to initialize `camInfoMsg` itself 🤣

```csharp
var camInfoMsg = new sensor_msgs.msg.CameraInfo();
camInfoMsg.K[0] = fx;
camInfoMsg.K[2] = cx;
camInfoMsg.K[4] = fy;
camInfoMsg.K[5] = cy;
camInfoMsg.K[8] = 1;
```

The above method is totally fine.

closed 2023-03-17 by chenjunnn

adamdbrw, 2023-03-17:
Glad this works for you! Different memory models of C# and C++ can be confusing and require some design choices when mixing :D