RobotecAI/ros2-for-unity #71: Custom message problem

dsponer, 2023-05-24:
Hello

I have a problem with my custom message

RuntimeError: type support not from this implementation, at /tmp/binarydeb/ros-foxy-rmw-fastrtps-cpp-1.3.1/src/subscription.cpp:91, at /tmp/binarydeb/ros-foxy-rcl-1.1.14/src/rcl/subscription.c:168
ROS2.Utils.CheckReturnEnum (System.Int32 ret) (at <1afb0f4c40de4f0192212174d671b2ca>:0)
ROS2.Subscription`1[T]..ctor (System.String subTopic, ROS2.Node node, System.Action`1[T] cb, ROS2.QualityOfServiceProfile qos) (at <1afb0f4c40de4f0192212174d671b2ca>:0)
ROS2.Node.CreateSubscription[T] (System.String topic, System.Action`1[T] callback, ROS2.QualityOfServiceProfile qos) (at <1afb0f4c40de4f0192212174d671b2ca>:0)
ROS2.ROS2Node.CreateSubscription[T] (System.String topicName, System.Action`1[T] callback, ROS2.QualityOfServiceProfile qos) (at Assets/Ros2ForUnity/Scripts/ROS2Node.cs:92)
ROS2.ROS2Light.Update () (at Assets/Ros2ForUnity/Scripts/ROS2Light.cs:53)

Custom messages were successfully collected. The C# script sees these messages.
But at the node startup stage, an error crashes

A custom message consists of two fields
uint8 chennel
uint8 brightness

My configuration
Ubuntu 20.04
ROS Foxy
.NET 3.1
ROS2Unity: foxy_fixes_for_custom_messages

pijaro, 2023-06-16:
`foxy_fixes_for_custom_messages` is some old branch of Ros2ForUnity. Can you try to build it with a stable branch like [1.2.0](https://github.com/RobotecAI/ros2-for-unity/tree/1.2.0)?

Did you take the whole asset (R2FU with your custom messages) from the `install` directory or did you mix your msgs libs with some other release?

Is it standalone or overlay mode? You might also be mixing the build and sourced version of ROS 2. Make sure you are running standalone without ROS 2 sourced or while using overlay - use the same ROS 2 distro.

izakisyoutarou, 2023-06-22:
Sorry, I am getting the same error.
I got the directory from install/asset.
The build was built in standalone mode.
I am using glactic as distribution.