RobotecAI/ros2-for-unity #72: some question about building with custom messages in standalone mode

Hezhexi2002, 2023-06-09:
@pijaro Hi,sorry to bother you here,but I meet some problem when I attempt to build it with my custom messages,the output of `./build.sh --standalone --clean-install` is shown below:
```
Summary: 23 packages finished [5min 7s]
  1 package failed: auto_aim_interfaces
  7 packages aborted: diagnostic_msgs nav_msgs sensor_msgs shape_msgs tf2 tf2_msgs trajectory_msgs
  22 packages had stderr output: action_msgs actionlib_msgs auto_aim_interfaces builtin_interfaces composition_interfaces diagnostic_msgs example_interfaces geometry_msgs lifecycle_msgs nav_msgs rcl_interfaces ros2cs_core rosgraph_msgs sensor_msgs shape_msgs statistics_msgs std_msgs std_srvs test_msgs tf2_msgs trajectory_msgs unique_identifier_msgs
  18 packages not processed
Ros2cs build failed!
```
as you can see the custom message is auto_aim_interfaces which is failed during building and here is the message in auto_aim_interfaces's stderr.log：
```
[0mCMake Error: The source directory "/root/ros2-for-unity/src/ros2cs/custom_messages/rm_auto_aim/auto_aim_interfaces" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.[0m
gmake: *** [Makefile:4039: cmake_check_build_system] Error 1
```
I'm guessing that the source directory `/root/ros2-for-unity/src/ros2cs/custom_messages/rm_auto_aim/auto_aim_interfaces`  does not exist is because I used to set the remote repo in ros2_for_unity_custom_messages.repos but it should not happen because now I just put the auto_aim_interfaces next to other packages in the src/ros2 sub-folder.Actually the package is needed by the project [rm_vision_simulator](https://github.com/chenjunnn/rm_vision_simulator) which is originally using [ros2-for-unity 1.2.0 release](https://github.com/RobotecAI/ros2-for-unity/releases/download/1.2.0/Ros2ForUnity_humble_windows10.zip),However,I want to build the ros2-for-unity in standalone mode in order to run the simulator without ros2 installation of the host.I'm not sure if I only need to build ros2-for-unity with auto_aim_interfaces or the entire [rm_auto_aim](https://github.com/chenjunnn/rm_auto_aim),hope you can give me some advice!:-)

pijaro, 2023-06-16:
Putting the message package manually should work fine :thinking: 

Out of curiosity, if you rerun the build command, does it fix the problem or is it persistent?

Hezhexi2002, 2023-06-17:
> Putting the message package manually should work fine 🤔
> 
> Out of curiosity, if you rerun the build command, does it fix the problem or is it persistent?

sorry to response so lately,actually it will still give the error message even when I rerun the command，but the author of [rm_vision_simulator](https://github.com/chenjunnn/rm_vision_simulator) has told me that the repo doesn't need to build the ros2-for-unity with custom message

Hezhexi2002, 2023-06-17:
> sorry to response so lately,actually it will still give the error message even when I rerun the command，but the author of [rm_vision_simulator](https://github.com/chenjunnn/rm_vision_simulator) has [...]

the error still exists even if I remove the custom message to build:
[image]

pijaro, 2023-06-28:
Hmm, I can't see any errors on the provided screenshot. :thinking: 

Some packages had a **stderr output**, but that doesn't mean that there are actual errors. The build system just uses stderr output to print some warnings at some point. You can ignore it.