RobotecAI/ros2-for-unity #75: ros2-for-unityでカスタムメッセージを実行したときにunity側でエラーが発生

izakisyoutarou, 2023-06-21:
はじめまして、ros2-for-unityのカスタムメッセージを実行しようとしたのですがuntiy側でエラーが発生してしまいました。
カスタムメッセージの作り方にはファイルをドラックアンドドロップするとできると書いてありますが、ROS2TalkerExample.csなどのファイルは書き換えなくて良いのでしょうか？
カスタムメッセージをどのようにしてパブリッシュするか知りたいです。
untiy側のエラー
Assets/Ros2ForUnity/Scripts/ROS2TalkerExample.cs(28,48): error CS0426: 'String' という型名が 'TwoInts' という型に存在しません。

追加したカスタムメッセージファイル
injection_interface_msg
my_messages

[injection_interface_msg.zip](https://github.com/RobotecAI/ros2-for-unity/files/11816069/injection_interface_msg.zip)
[my_messages.zip](https://github.com/RobotecAI/ro2-for-unity/files/11816081/my_messages.zip)
[asset.zip](https://github.com/RobotecAI/ros2-for-unity/files/11816108/asset.zip)

よろしくお願いします。

pijaro, 2023-06-28:
Hello,

No, you do not have to rewrite any files to build and use your custom messages.

The error you describing doesn't look like connected with custom messages. Did this error happen after you modified the `ROS2TalkerExample.cs` script or before any modifications? 

`TwoInts` type is connected with services (`ROS2ServiceExample.cs`), therefore it should not be a part of a talker example (topic publish example).

izakisyoutarou, 2023-07-03:
Thank you. I was able to solve the problem, but when I tried to run the program on the untiy side, I got an error on the unity side. I would like to know how to deal with this problem. Thank you very much.
RuntimeError: type support not from this implementation, at /tmp/binarydeb/ros-foxy-rmw-fastrtps-cpp-1.3.1/src/publisher.cpp:86, at /tmp/ binarydeb/ros-foxy-rcl-1.1.14/src/rcl/publisher.c:180
ROS2.Utils.CheckReturnEnum (System.Int32 ret) (at <680ec49a722e41208641c44cc0cd291a>:0)
ROS2.Publisher`1[T]. .ctor (System.String pubTopic, ROS2.Node node, ROS2.QualityOfServiceProfile qos) (at <680ec49a722e41208641c44cc0cd291a>:0)
ROS2.Node.CreatePublisher[T] (System.String topic, ROS2.QualityOfServiceProfile qos) (at <680ec49a722e41208641c44cc0cd291a>:0)
ROS2.ROS2Node.CreatePublisher[T] (System.String topicName, ROS2.QualityOfServiceProfile qos) (at Assets/Ros2ForUnity/Scripts/ROS2Node.cs :75)
ROS2.ROS2TalkerExample.Update () (at Assets/Ros2ForUnity/Scripts/ROS2TalkerExample.cs:43)

pijaro, 2023-07-19:
Maybe the fastdds RMW was missing during the compilation? 

Try installing it `sudo apt install ros-foxy-rmw-fastrtps-cpp` and **clean** recompile the Ros2ForUnity. 

I don't guarantee it will work, since we don't support `foxy` anymore.