RobotecAI/ros2-for-unity #76: Talker example does not seem to work with "normal" ROS2.

samiamlabs, 2023-06-26:
**Describe the bug**
Talker/listener example seems to work between different instances of Unity, but not with normal (or docker based) ROS2.

**To Reproduce**
Steps to reproduce the behavior:
1. Download https://github.com/RobotecAI/ros2-for-unity/releases/download/1.3.0/Ros2ForUnity_humble_standalone_ubuntu22.zip and add it to an empty Unity project.
2. Add ROS2 Unity Component and ROS2 Talker Example to a new GameObject.
3. Open termainal with ROS2 humble (native or in docker) and try to list or echo the "/chatter" topic.
4. See error

**Expected behavior**
Topics published from Unity should be accessable from "normal" ROS2.

**Screenshots**
[image]

**Desktop:**
 - OS: [e.g. Ubuntu20204
 - ros2 distro [humble]
 - ros2-for-unity version [1.3.0]
 - ros2 dds middleware [fastdds]
 - ros2 environment setup [single pc]

pijaro, 2023-06-28:
I can't reproduce it.

How do you run your editor? If you are using the Unity Hub launcher, then your .bashrc is not sourced in the editor. 

If you are setting the ROS_DOMAIN or other ROS 2 settings in bashrc, then you might have a mismatched configuration between terminal and the editor.

samiamlabs, 2023-06-29:
It started to work temporarily on the ROS2 installation on my computer after about an hour for some reason, so it is not ROS_DOMAIN_ID or anything like that. Thinking it has something to do with Fast DDS autodiscovery. Will look into it more.

samiamlabs, 2023-08-22:
After a fresh reinstall of Ubuntu and ROS2 humble, communication with local ROS2 now works fine for me. Strange...
Don't have a good theory for why it was not working before. I didn't have any custom DDS settings or anything like that.

I found a similar problem with ROS2 communication to docker though.
You should be able to recreate it by running the talker example in Unity and:
```bash
docker run --network host -it osrf/ros:humble-desktop
ros2 topic echo /chatter
```

Switching to from `rmw_fastrtps_cpp` to `rmw_cyclonedds_cpp` in Unity fixed this for me.
```
Environment.SetEnvironmentVariable("RMW_IMPLEMENTATION", "rmw_cyclonedds_cpp");
```

In my experience, autodiscovery in humble works much better with cyclone at the moment.
Maybe it should be the default, or at least it would be nice if there was an easy way to select it as a user.
A scriptable object with fields for some of the standard environment variables, like ROS_DOMAIN_ID, perhaps?

pijaro, 2023-08-23:
Yes, I've noticed that, for some reason, `fastrtps` struggles in standalone mode, but works well with "loopback" scenario (sub and pub are both inside the Unity). Strangely, there are no errors whatsoever. We need to investigate it deeper :thinking:.

However, there are no issues with `cyclonedds`, therefore, for now, I suggest using that.

FelipeH92, 2024-02-28:
Hi, any news on the fastrtps problem? Here I can execute a talker and listener nodes in Unity, and echo them in a terminal on the same machine, but in a different machine I can't see the nodes and topics in Unity (but I can see the nodes and topics I execute in powershell in that same pc), so from my second PC I can any nodes and topics outside unity on the first pc, but not the unity ones, and in powershell on my first pc I can see both the Unity nodes and the second pc nodes.

s-m-asjad, 2024-06-21:
@FelipeH92 I was running into the same issue. The solution was to use [husarnet](https://husarnet.com/blog/husarnet-dds/). Although completely unrelated but we were teleoperating robots across different networks and had to be using it for the project. 

It turns out that if we turned husarnet off and tried replicating our setup locally with Unity, we were running into the exact same issue you described i.e.

- being able to send and receive ROS2 messages from a remote machine to a local machine but unable to see them in Unity
- being able to send and receive ROS2 messages from Unity to a local machine but unable to see them on the remote machine.

We tried republishing topics using the `topic_tools` package on the windows machine but that is obviously not a scalable solution.