RobotecAI/ros2-for-unity #78: Standalone asset has extra dependencies

neilharrison, 2023-07-05:
On a clean install of Ubuntu 22.04 with no ROS2 installed, the Unity asset built using the docker image doesn't work directly. There are two packages that need to be installed to get it work correctly - `libspdlog-dev` , `libtinyxml2-dev`

Without  `libspdlog-dev` - the following error shows in Unity with the talker and listener examples:

> UnsatisfiedLinkError: librcl.so
> ROS2.DllLoadUtilsUnix.Load (System.String libraryFileName) (at <a2a919017a70434a94efb7733ae66da5>:0)
> ROS2.DllLoadUtilsUnix.LoadLibraryByName (System.String libraryFileName) (at <a2a919017a70434a94efb7733ae66da5>:0)
> ROS2.DllLoadUtilsUnix.LoadLibraryNoSuffix (System.String fileName) (at <a2a919017a70434a94efb7733ae66da5>:0)
> ROS2.NativeRcl..cctor () (at <680ec49a722e41208641c44cc0cd291a>:0)
> Rethrow as TypeInitializationException: The type initializer for 'ROS2.NativeRcl' threw an exception.
> ROS2.ROS2ForUnity..ctor () (at Assets/Ros2ForUnity/Scripts/ROS2ForUnity.cs:311)
> ROS2.ROS2UnityComponent.LazyConstruct () (at Assets/Ros2ForUnity/Scripts/ROS2UnityComponent.cs:61)
> ROS2.ROS2UnityComponent.Ok () (at Assets/Ros2ForUnity/Scripts/ROS2UnityComponent.cs:49)
> ROS2.ROS2TalkerExample.Update () (at Assets/Ros2ForUnity/Scripts/ROS2TalkerExample.cs:38)

And without  `libtinyxml2-dev` : 

> RuntimeError: failed to get symbol 'rmw_init_options_init' due to Environment variable 'AMENT_PREFIX_PATH' is not set or empty, at ./src/functions.cpp:171, at ./src/rcl/init_options.c:75
> ROS2.Utils.CheckReturnEnum (System.Int32 ret) (at <680ec49a722e41208641c44cc0cd291a>:0)
> ROS2.Ros2cs.Init () (at <680ec49a722e41208641c44cc0cd291a>:0)
> ROS2.ROS2ForUnity..ctor () (at Assets/Ros2ForUnity/Scripts/ROS2ForUnity.cs:311)
> ROS2.ROS2UnityComponent.LazyConstruct () (at Assets/Ros2ForUnity/Scripts/ROS2UnityComponent.cs:61)
> ROS2.ROS2UnityComponent.Ok () (at Assets/Ros2ForUnity/Scripts/ROS2UnityComponent.cs:49)
> ROS2.ROS2TalkerExample.Update () (at Assets/Ros2ForUnity/Scripts/ROS2TalkerExample.cs:38)

Neither of these errors seem to have any relation to the packages that are missing and the second one looks like an issue with rcl rather than ros2cs, however once both of these packages are installed the standalone asset works perfectly!

Setup:
 - OS: Ubuntu 22.04
 - ros2 distro: humble
 - ros2-for-unity version:  git-sha a195bf6a2ed776e90ab89161273712419d640f85
 - ros2cs version: git-sha 1edbed143ae6ec806d43350330f67fd2c40a45de
 - ros2 dds middleware : fastdds
 - ros2 environment setup: not setup locally, only used within docker containers

pijaro, 2023-07-07:
Thanks, will look into it! :eyes:

pijaro, 2023-07-07:
Fix is work in progress ([ros2cs branch](https://github.com/RobotecAI/ros2cs/tree/fix/standalone_missing_dependencies)). It works for humble and u22, but I need to do some more tests on other OSes and configurations :+1:

pijaro, 2023-07-19:
Should be fixed by https://github.com/RobotecAI/ros2cs/pull/54 (now on `ros2cs` develop branch) :+1:

closed 2023-07-19 by pijaro