RobotecAI/ros2-for-unity #79: The message "Invalid message type" will appear.

izakisyoutarou, 2023-07-10:
I got the following error when using ros2-for-unity to publish a custom message. Please let me know how to deal with it.

[image]

publisher's source code
[image]

custommessage file
/home/izakisyotaro/ros2_studies_ws/src/my_messages.zip

version ubuntu 22.04
distribution humble
build ./build.sh --standalone

adamdbrw, 2023-07-10:
@izakisyoutarou this looks like you are successfully publishing your custom message, but receiving it is the problem. Did you source your custom package (my_messages) also in the terminal where you typed `ros2 topic echo`?

izakisyoutarou, 2023-07-10:
Thanks for your reply @adamdbrw
The custom messages (my_messages) only has a cmake file, an xml file, and a message file.
So I don't think I can do ros2 topic echo because there is no topic name, how do you do it?

adamdbrw, 2023-07-11:
You posted a screen where you ran `ros2 topic echo` command. In this console, source your built message package first. Please refer to ROS 2 resources (tutorials etc) on how to do that.

izakisyoutarou, 2023-07-12:
I forgot the source install setup bash command. I was able to run my own message successfully. Thank you very much for your support.

closed 2023-07-12 by adamdbrw

izakisyoutarou, 2023-07-12:
Sorry, you asked me to reply to this.

Closed #79<https://github.com/RobotecAI/ros2-for-unity/issues/79> as completed.