RobotecAI/ros2-for-unity #81: Android app installation with ros2-for-unity does not work

izakisyoutarou, 2023-07-26:
I'm sorry, I'm trying to install an app on android with ros2-for-unity and when I run build, I get the following error.
I would like to know how to solve this problem.
I went to "Player Settings..." and found the Scripting Backend section. I was able to install the app, but turtlesim did not work.
Does this mean that there is a problem with IL2CPP?
[image]

pijaro, 2023-07-27:
Sorry, but Ros2ForUnity isn't supported on Android devices :(

izakisyoutarou, 2023-07-27:
This URL seems to be supported, but does it mean it is not?

https://qiita.com/Kotakku/items/cdc3eca89dd8aec4ee86

pijaro, 2023-07-27:
We didn't prepare the R2FU to run on Android, but it looks like, indeed, someone from the community did. :+1: 

Looks like you can find the forks here: https://github.com/Kotakku?tab=repositories

As for your issue, are these 2 lines the whole error message? Is there something else after the `clang: not found` statement? Maybe it mentions what couldn't be found?

closed 2023-09-23 by izakisyoutarou