RobotecAI/ros2-for-unity #83: When I use the latest windows 11 standalone, my unity is stuck and unresponsive.

soul667, 2023-08-15:
When I use the latest windows 11 standalone, my unity is stuck and unresponsive.
Through my code modifications, I realized that the problem was with the node creation statement just like this.
```c#
 ros2Node = ros2Unity.CreateNode("ROS2UnityTalkerNode");
```
I then consulted the unity log `Editor.log` and found the following error
```
Fallback handler could not load library C:/Program Files/Unity/Hub/Editor/2023.1.7f1c1/Editor/Data/MonoBleedingEdge/lib/libdl.so.2
Fallback handler could not load library C:/Program Files/Unity/Hub/Editor/2023.1.7f1c1/Editor/Data/MonoBleedingEdge/lib/libdl.so.2.dll
Fallback handler could not load library C:/Program Files/Unity/Hub/Editor/2023.1.7f1c1/Editor/Data/MonoBleedingEdge/lib/libdl.so.2.dll
Fallback handler could not load library C:/Program Files/Unity/Hub/Editor/2023.1.7f1c1/Editor/Data/MonoBleedingEdge/lib/libdl.so.2
Fallback handler could not load library C:/Program Files/Unity/Hub/Editor/2023.1.7f1c1/Editor/Data/MonoBleedingEdge/lib/libdl.so.2
Fallback handler could not load library C:/Program Files/Unity/Hub/Editor/2023.1.7f1c1/Editor/Data/MonoBleedingEdge/lib/libdl.so.2.dll
Fallback handler could not load library C:/Program Files/Unity/Hub/Editor/2023.1.7f1c1/Editor/Data/MonoBleedingEdge/lib/libdl.so.2.dll
Fallback handler could not load library C:/Program Files/Unity/Hub/Editor/2023.1.7f1c1/Editor/Data/MonoBleedingEdge/lib/libdl.so.2
Fallback handler could not load library C:/Program Files/Unity/Hub/Editor/2023.1.7f1c1/Editor/Data/MonoBleedingEdge/lib/libdl.dylib
Fallback handler could not load library C:/Program Files/Unity/Hub/Editor/2023.1.7f1c1/Editor/Data/MonoBleedingEdge/lib/libdl.dylib.dll
Fallback handler could not load library C:/Program Files/Unity/Hub/Editor/2023.1.7f1c1/Editor/Data/MonoBleedingEdge/lib/libdl.dylib.dll
Fallback handler could not load library C:/Program Files/Unity/Hub/Editor/2023.1.7f1c1/Editor/Data/MonoBleedingEdge/lib/libdl.dylib
Fallback handler could not load library C:/Program Files/Unity/Hub/Editor/2023.1.7f1c1/Editor/Data/MonoBleedingEdge/lib/libdl.dylib
Fallback handler could not load library C:/Program Files/Unity/Hub/Editor/2023.1.7f1c1/Editor/Data/MonoBleedingEdge/lib/libdl.dylib.dll
Fallback handler could not load library C:/Program Files/Unity/Hub/Editor/2023.1.7f1c1/Editor/Data/MonoBleedingEdge/lib/libdl.dylib.dll
Fallback handler could not load library C:/Program Files/Unity/Hub/Editor/2023.1.7f1c1/Editor/Data/MonoBleedingEdge/lib/libdl.dylib
ROS2 version: humble. Build type: standalone. RMW: rmw_fastrtps_cpp
UnityEngine.StackTraceUtility:ExtractStackTrace ()
UnityEngine.DebugLogHandler:Internal_Log (UnityEngine.LogType,UnityEngine.LogOption,string,UnityEngine.Object)
UnityEngine.DebugLogHandler:LogFormat (UnityEngine.LogType,UnityEngine.Object,string,object[])
UnityEngine.Logger:Log (UnityEngine.LogType,object)
UnityEngine.Debug:Log (object)
ROS2.ROS2ForUnity:.ctor () (at Assets/Ros2ForUnity/Scripts/ROS2ForUnity.cs:316)
ROS2.ROS2UnityComponent:LazyConstruct () (at Assets/Ros2ForUnity/Scripts/ROS2UnityComponent.cs:58)
ROS2.ROS2UnityComponent:Start () (at Assets/Ros2ForUnity/Scripts/ROS2UnityComponent.cs:68)

(Filename: Assets/Ros2ForUnity/Scripts/ROS2ForUnity.cs Line: 316)

``

Deric-W, 2023-08-16:
I had a similar problem, check if [clearing `C:\ProgramData\eprosima\fastrtps_interprocess`](https://answers.ros.org/question/402008/ros2-windows-node-always-hang) fixed your problem.

soul667, 2023-08-19:
thank you so much,it,s really solved my problem.

soul667, 2023-08-19:
all is done

closed 2023-08-19 by soul667