RobotecAI/ros2-for-unity #84: failed to get symbol 'rmw_init_options_init'

samiamlabs, 2023-08-21:
**Describe the bug**
The standalone 1.3 release for humble/ubuntu 22.04 fails to get a symbol when running the `Ros2cs.Init()` function.

**To Reproduce**
Steps to reproduce the behavior:
1. Start with a clean Ubuntu 22.04 install
2. Do not install ROS2 (not sure if this is relevant)
3. Create a new Unity project with editor version 2022.3.7
4. Install `libspdlog-dev` using apt-get
5. Download and unzip https://github.com/RobotecAI/ros2-for-unity/releases/download/1.3.0/Ros2ForUnity_humble_standalone_ubuntu22.zip
6. Move the unzipped folder under Assets in the Unity project.
7. Add the ROS2UnityComponent script to a game object in the scene.
8. Press play
9. See error

The full error is:
```
RuntimeError: failed to get symbol 'rmw_init_options_init' due to Environment variable 'AMENT_PREFIX_PATH' is not set or empty, at ./src/functions.cpp:171, at ./src/rcl/init_options.c:75
```

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [Ubuntu22.04]
 - ros2 distro [humble]
 - ros2-for-unity version [1.3.0]
 - ros2 environment setup [single pc]

samiamlabs, 2023-08-22:
I installed ROS2 locally on the same computer with the error, and it disappeared.

It would be really nice to have this working on computers without needing ROS2 installed.
We pretty much exclusively use ROS2 in docker at my company and don't tend to have a local ROS2 install.

pijaro, 2023-08-23:
Hi, I believe you were missing `libtinyxml2-dev` as well.

This should be [fixed now](https://github.com/RobotecAI/ros2cs/pull/54) (currently on `develop` ros2cs branch).