RobotecAI/ros2-for-unity #92: Is it possible to set a ROS DOMAIN ID?

izakisyoutarou, 2023-11-03:
I would like to set up a ROS DOMAIN ID when using ros2-for-unity.
I would like to know if it is possible to set a ROS DOMAIN ID for ros2-for-unity.
If so, how should I write the script?
Thank you in advance for your reply.
Version ubuntu 22.04
Distribution humble

antoniskara, 2023-11-10:
That would be really usefull

izakisyoutarou, 2023-11-11:
Resolved, it is now possible to declare the DOMAINID in an environment variable before the start function using the awake function. Thank you very much.

closed 2023-11-11 by izakisyoutarou

jvmoraiscb, 2023-11-17:
Could you send the code snippet of how you did that?

izakisyoutarou, 2023-11-20:
@jvmoraiscb 
Sorry, it took me a while to reply.
Here is the code.
I used the awake function and Environment.SetEnvironmentVariable to set it up.
The DOMAIN ID at this time is 42.
ROS2ListenerExample can be set the same way
[image]

lucianoreul, 2024-04-15:
I'm working with Windows, and it's not working when I run this script. Any suggestions?

When you define environment variables from a command line, the values don’t persist across sessions. I think its because the unity Hub opens the Unity as a child session. I tried some options how to run a cmd command to update terminal session, but not working.

jvmoraiscb, 2024-04-16:
First of all, I'm using overlay mode. I couldn't do this using standalone mode.

In the first terminal, I use this command to set the global environment variable for future sessions:
```bash
setx /m ROS_DOMAIN_ID 42
```
Then, in a new terminal (powershell), I load the ROS variables and start a process by directly opening the Unity executable (I also had issues opening the hub) and pass the path to my project folder as a parameter:
```bash
C:\dev\ros2_humble\local_setup.ps1 ; Start-Process -FilePath '<path\to\unity>' -ArgumentList '-projectPath "<path\to\unity-project>"'
```

If you have any further questions, I communicate much better in Portuguese 😄 .

Feel free to send an email to jv.moraiscb@gmail.com