RobotecAI/ros2-for-unity #99: does ros2-for-unity support ros2 iron?

yangming517, 2024-03-11:
In the environment of ROS2 humble version on Ubuntu 22.04, I have successfully run the ros2-for-unity plugin, but I found some defects in ros2 humble. So, I want to replace ROS2 humble with the Iron version in the Ubuntu 22.04 environment. Are there any friends who have successfully run ros2-for-unity in the ubuntu 22.04 ros2 iron environment? Please tell me if this is feasible?thank you very much .

Muthukumar4796, 2024-04-13:
How did your experience with ROS2 Humble in Unity go? I installed Unity on my Ubuntu 22 and downloaded the ros2 humble unity package, placing it in the assets folder. What should I do next? Also, what defects did you face?

adamdbrw, 2024-04-24:
We don't have an Iron release, but you can try building it from source.

yangming517, 2024-06-23:
> How did your experience with ROS2 Humble in Unity go? I installed Unity on my Ubuntu 22 and downloaded the ros2 humble unity package, placing it in the assets folder. What should I do next? Also, what defects did you face?
I used overlay mode,which means that ros2 should be install in your ubuntu.I haven't try standalone mode.

yangming517, 2024-06-23:
> How did your experience with ROS2 Humble in Unity go? I installed Unity on my Ubuntu 22 and downloaded the ros2 humble unity package, placing it in the assets folder. What should I do next? Also, what defects did you face?

The defect is about moveit2 in ros2,Humble doesn't have full function of moveit2.

yangming517, 2024-06-23:
> We don't have an Iron release, but you can try building it from source.

ok,thank you