RobotecAI/ros2-for-unity #100: Issue to change RMW implementation

manfer33, 2024-03-19:
I have one computer with Windows11 Unity running R2ForUHumble standalone version and another computer with Ubuntu 22 and ROS2 Humble. Both on the same local network. Humble in Ubuntu with default setup.

The computer with Windows11 works fine but the computer with Ubuntu can not see the topics from the Windows computer. I read in another posts similar issues related with FASTDDS so I tried to change it for Cyclone, but it doesn't work.

I am using the command that I saw from other posts, but no one says were to place it:
`Environment.SetEnvironmentVariable("RMW_IMPLEMENTATION", "rmw_cyclonedds_cpp");
`

I tried to check another standalone distro but 1.3.0 is only for humble in Windows and previous versions are for Windows 10

Shunichi09, 2024-04-23:
I would like to change the RMW implementation, too.
Has anyone solve this problem??

adamdbrw, 2024-04-24:
Setting Widnows environmental variable RMW_IMPLEMENTATION should work. See https://docs.ros.org/en/rolling/Tutorials/Beginner-CLI-Tools/Configuring-ROS2-Environment.html.

Shunichi09, 2024-04-27:
Thank you for your response!
Ok, I'll try that!

sdyby2006, 2024-05-16:
> Thank you for your response! Ok, I'll try that!

Hello, can it be successfully modified? By changing environmental variables and executing `Environment.SetEnvironmentVariable("RMW_IMPLEMENTATION", "rmw_cyclonedds_cpp"); `it still doesn't take effect. Upon startup, it still shows `ROS2 version: humble. Build type: standalone. RMW: rmw_fastrtps_cpp.` Thank you for your help.

adamdbrw, 2024-05-16:
The API you use is too late on linux since the process is already running, you can look at how ros2 does the check in rclcpputils IIRC. In short,  this way of setting environment variable is not effective to modify RMW_IMPLEMENTATION for ROS. Instead, set it in your bash/profile/system before running R2FU app.