RobotecAI/ros2-for-unity #102: Can't add custom messages to my Unity Project

MathisFeDaSi, 2024-04-22:
Hi there !
I've firstly tried your ROS2 package for Unity and it works perfectly !! Thanks for your work :D
Then I needed some custom messages/services (you can find my repository [here](https://github.com/MathisFeDaSi/CartoBot_Interfaces.git)), so I added them in the [ros2_for_unity_custom_messages.repos](https://github.com/RobotecAI/ros2-for-unity/files/15063609/ros2_for_unity_custom_messages.repos.txt) but it doesn't work: *The type or namespace name 'pointer_interfaces' could not be found (are you missing a using directive or an assembly reference?)* (pointer_interfaces is one of my packages)
I can build the files, but I can't create a .unitypackage: [log.txt](https://github.com/RobotecAI/ros2-for-unity/files/15063679/log.txt)
And adding them manually to the Unity project don't throw a compiler error, it states it can't find them when using them.

Any help would be greatly appreciated,
Best regards,
Mathis

MathisFeDaSi, 2024-04-22:
PS:
I'm on **Ubuntu 22.04**
With the **Humble** distro
And **Unity 2022.3.22f1** version

adamdbrw, 2024-04-22:
I noticed you are using a custom message for a 2D point, maybe consider standard one: https://docs.ros2.org/latest/api/geometry_msgs/msg/Point32.html (keep z as 0).

MathisFeDaSi, 2024-04-22:
Thanks for your insane reactivity !!
I will try that right now ;)
I'm keeping you in touch

And about the std_point message it's because I will surely add some information further but thanks for letting me know :D

MathisFeDaSi, 2024-04-22:
I've tried your way, by adding directly my two packages with custom message inside the src/ros2cs/custom_messages folder.
The build worked, the .unitypackage no and so in Unity :'(

I don't know if you have any idea about that, but I think I will remake it from the beginning once more (I've done it all day long :'D ) but can you pls guide me on how to properly create my repository for my custom messages. Knowing that this repo will only contain custom messages for a unique project. I don't know if it's better to make different packages or put anything in only one ?
And what I have to put in this/these repo(s) pls ?

I would prefer to do it by using repo(s) because we are 3 to work on this project

I know it could seems like I haven't read the ream.me at all but I promise I've walk through all the steps in both ways several times I'm really stuck on this :(
But once it will work (and I hope soon with your help <3 ) I would add some information/example to the read.me to avoid people being stuck like me (more information can't be disturbing ;) )

Thank you :D
Mathis

MathisFeDaSi, 2024-04-24:
Hey!
I've solved my issue :D
I didn't have git-lfs installed, so ros2cs couldn't build normally but didn't return an error as I was building R2FU.
I think that adding a simple line in the prerequisites saying to install git-lfs could be useful for some people ;)
I noticed it by attempting to manually build only ros2cs and here the error telling me I didn't have git-lfs showed up.

So anything linked to custom messages :')
(I don't know if you can rename this issue as "can't build repository")

And I use a GitHub repo for referencing my custom messages, that works perfectly!
[ros2_for_unity_custom_messages.repos.txt](https://github.com/RobotecAI/ros2-for-unity/files/15096292/ros2_for_unity_custom_messages.repos.txt) (you can check my repo it's public) (You can check my repo, it's public)

Thank you, and I hope it would help someone else :v:
Cheers!

closed 2024-04-24 by MathisFeDaSi