RobotecAI/ros2-for-unity #103: Connection problem between ROS2 in WSL2 and Unity in Windows

Shunichi09, 2024-04-23:
Hi! Thank you for your great work.

I'm trying to build a system which includes Unity in Windows11 and ROS2 in WSL2.

However,  I can not pub/sub between ROS2 in WSL2 and Unity in Windows.

Is there anything I have to 
write in wslconfig??
I think the problem is Networking between WSL2 and Windows.
Are there anyone who tried this setting??

I'm sorry I don't know where to write, so I decided to write this as a Issue.

Thank you for your help.

Shunichi09, 2024-04-23:
Ok, now I understand the problem is caused by ROS DOMAIN ID.

In ros2-for-unity, how I define ROS DOMAIN ID ??

Shunichi09, 2024-04-23:
This solution did not work for me...
I'm using standalone Windows re-build pkg.

- https://github.com/RobotecAI/ros2-for-unity/issues/92

adamdbrw, 2024-04-24:
Setting ROS_DOMAIN_ID on Linux will work, I have no idea about Windows or WSL2. We don't test or support WSL2 unfortunately.

Shunichi09, 2024-04-25:
Thank you for your response!
Ok! I understand.

closed 2024-04-25 by Shunichi09