RobotecAI/ros2-for-unity #108: communication with ROS2 and custom message build procedure

sawan-kcl, 2024-06-05:
Hello,
I recently installed this package and everything seems to be working correctly but the topic being published/subscribed in the example script do not seem to be detectable by my local ROS2 installation.
Did i miss any steps?

Few information that might be useful:

I'm using Ubuntu 22.04 with ROS2 humble.
* I have sourced ROS2 in my `~/.bashrc` and `~/.profile`
* I dont' have any external connection set up (Like unity-robotics-hub) in the current project in which I'm testing this.
* I installed in --standalone mode so not starting unity editor from inside a ROS sourced terminal. Just starting it normally from launcher shortcut.

Also, please could you tell me what is the procedure to include and use custom ROS2 messages that reside in local repository in my ROS2 workspace?

s-m-asjad, 2024-06-21:
Can you try adding `Application.runInBackground = true;` inside your `void start()` ?

 I had similar issues on Windows with Unity - ROS2 Humble.

s-m-asjad, 2024-06-21:
For custom messages, I am not sure if the procedure is different but I have been following [these steps](https://answers.ros.org/question/403535/ros2-for-unity-create-custom-msg/)  to add a custom message on Windows. We have to rebuild the ROS2 Unity asset each time I add a new custom message.

Hsaisai, 2024-11-26:
> For custom messages, I am not sure if the procedure is different but I have been following [these steps](https://answers.ros.org/question/403535/ros2-for-unity-create-custom-msg/) to add a custom message on Windows. We have to rebuild the ROS2 Unity asset each time I add a new custom message.

hello, can you teach me how to add custom messages?What is the writing format?Thanl you very much!

s-m-asjad, 2024-11-26:
@Hsaisai are you asking about adding a custom message in ROS2 on Ubuntu, ROS2 on Windows, or ROS2 inside Unity on Windows?

Hsaisai, 2024-11-27:
> @Hsaisai are you asking about adding a custom message in ROS2 on Ubuntu, ROS2 on Windows, or ROS2 inside Unity on Windows?

ROS2 inside Unity on Ubuntu.