RobotecAI/ros2-for-unity #109: How to solve 'UnsatisfiedLinkError: librcl.so'?

phs4530, 2024-06-07:
- Ubuntu 20.04
- Unity 2020.3.6f1

Hello! Thank you for reading my question. 
I am planning to use ros2 using the strongalone version. However, the following error occurs. Is there a solution?
Thank you for your reply in advance

--
[image]
--
[image]

omerts, 2024-06-20:
This is really a show stopper for us, if anyone can help solve this, please update

adamdbrw, 2024-07-01:
It looks like your ROS distro is not visible. In case of standalone version, ROS libraries should be shipped with the Unity package.
Are you building from source, or using a release package?

omerts, 2024-07-04:
@adamdbrw I used: [Ros2ForUnity_humble_standalone_ubuntu22.zip](https://github.com/RobotecAI/ros2-for-unity/releases/download/1.3.0/Ros2ForUnity_humble_standalone_ubuntu22.zip)

adamdbrw, 2024-07-06:
Unfortunately, our support for the project is very limited at the moment, as it just works for one project we use it for and all the other projects are on Open 3D Engine, much recommended as free Unity3D alternative

adamdbrw, 2024-07-06:
Perhaps @pijaro could help you as he prepared the build

pijaro, 2024-07-15:
The second error may be connected to the issue (calling get_data from non-main thread).

Can you confirm, that running everything from the main thread still throws the ROS 2 error?

phs4530, 2024-07-25:
@omerts I found it resolved in the latest branch of ros2cs.
- https://github.com/RobotecAI/ros2-for-unity/issues/78
However, the distribution version has not been updated yet.

closed 2024-07-25 by phs4530