RobotecAI/ros2-for-unity #110: Cannot use custom messages in unity

shwwoooo, 2024-06-12:
Hi, I am on Ubuntu 22.04 and using ROS humble. I built ros2-for-unity with custom messages by copying the messages source directories (the .msg and cmake files) to `src/ros2cs/custom_messages`. ros2-for-unity can be built, but when I copy the asset over to unity workspace and try to use the custom messages, unity complains `'GimbalCmd' does not contain a definition for 'position'`. Here are my code in c# and msg files.
subscriber.cs
```
transform.localRotation = Quaternion.Euler(msg.position.pitch, msg.position.yaw, 0.0f);
```
GimbalCmd.msg
```
# control dada
Gimbal position
Gimbal velocity
```
Gimbal.msg
```
float32 yaw
float32 pitch
```
Can anyone give some pointers in how to use custom messages? Thanks!

adamdbrw, 2024-06-12:
Please check the generated messages. C# convention for field naming uses uppercase letters. msg.Position should work.

shwwoooo, 2024-06-12:
Thank you for your quick response! Yes it works

closed 2024-06-12 by shwwoooo

robotasun, 2025-04-09:
> Please check the generated messages. C# convention for field naming uses uppercase letters. msg.Position should work.

Where can I find the generated messages for ros2cs? I've looked into build but they all had the lowercase standard