RobotecAI/ros2-for-unity #116: Transfer data from Unity to another computer with ROS2

L1MSpace, 2024-09-13:
I have a goal, to create a connection between two computers. The first computer on Windows 11 has Unity and ROS2 Humble installed. The second computer on Linux (Ubuntu 22.04) has only ROS2 Humble installed.

I was able to run the ros2-for-unity scripts “ROS2 Unity Component”, “ROS2 Talker Example”, “ROS2 Listener Example” in Unity without errors. Afterward, when I type “ros2 topic list” in the console on Windows, I see the “/chatter” topix that Unity creates. But, if I enter the same thing on a second computer on Linux, the “/chatter” topix is not there, i.e. ROS2 on Linux does not see this topix.
Both computers are connected to the same WI-FI network.

Question, does ros2-for-unity support data transfer via topics to another computer with ROS2? 
If not, are there then alternative libraries that are suitable for my purpose.

P.S. I used ROS# library for the same purpose in ROS1. There I just had to put in the ip of a second computer on Linux (where ros-master was running), and after that Unity could send and receive data from another computer with ROS2.

Environment:
Windows 11;
Linux (Ubuntu 22.04)
ROS2 Humble (on Windows and Linux);
ros2-for-unity 1.3.0, overlay mode.

JerryWu6288, 2025-02-06:
遇到了一样的问题，经测试他这个插件默认的DomainID应该是0，但好像没有地方能够修改这个id

Samarthraman01, 2025-02-25:
hey i have been facing the same issue, considering the fact that i was using the same setup as yours, but it is working with now.
just as stated previously make sure they both are in same ROS_DOMAIN_ID.
add the domain id in ros componenet script just to make sure it remains unchanged and same id in ubuntu system aswell.